jhemcu/FC-ESC-Firmware #3: Updating GHF405 AIO Pro 45A Fails - "Programming: FAILED" after verification.

francis916, 2021-12-20:
Hi. I tried to update the FC with the latest Betaflight firmware and it failed. I tried many other targets but they all failed and finally I tried the firmware downloaded from here. It also failed like below. I tried Mac(Intel & Apple silicone), and PC and changed cables. Any idea what I should do?

Thanks.

[image]

jhemcu, 2021-12-21:
主要还是看看驱动安装是否OK,正常写入我发给你的两个文件，肯定是可以的

董家良
津航电子

 联系我

francis916, 2021-12-21:
Hi. Thank you for your response. DFU connection is open successfully and it writes okay but it always fails at the end of 'Verification' process. I captured the video before. It's exactly the same with the firmware you provided. 

https://youtu.be/yf5sGOzi1rE

I mainly use Mac and it doesn't have the driver issue. I tried PC also and installed all the drivers and DFU opened on PC as well. It actually successfully erases the flash and also flashes. It's the verification process that it fails. 

Do you think it's the hardware issue? 

Thank.

jhemcu, 2021-12-21:
第一，有一处错误 ，GHF405AIO PRO 的固件是JHEF405PRO，而不是JHEF405.
第二，视频我看了，这种情况我们还没有遇到过，不过这种情况貌似和硬件没啥关系，我另外给你一套软件ST原厂的刷写固件的软件"DfuSe_Demo_V3.0.3_Setup.exe"，由于这个文件容量过大，邮件无法发送，你可以到ST网站上进行下载，你测试看看。用这个软件需要加载附件里面“betaflight_4.2.8_STM32F405.dfu”，以DFU后缀的文件名。

董家良
津航电子

 联系我

 

 
https://youtu.be/yf5sGOzi1rE

francis916, 2021-12-21:
Thank you! I'll try the method and report.

Best,

SH

francis916, 2021-12-21:
I have tried a few things. 

- I downloaded DfuSe Demo 3.0.6 because only the latest one was available from ST site. But it didn't detect DFU (screen shot 1) on Windows 10. I saw that it's replaced by STM32CubeProgrammer so I installed it. 

[image]

- STM32Cube detected and connected to the FC. It read the flash memory and it's all filled with 'FFFF'. I guess it's erased. 
[image]

- I loaded the firmware file from this GitHub and it has different content than the flash memory. 
[image]

- I tried to download the firmware to the flash memory. It seemed it's going well but the verification failed. I disconnected and reconnected the FC and saw the flash memory wasn't changed - it's still filled with 'FFFF'. 
[image]

- I tried checking 'Read Unprotect (MCU)' but  STM32Cube couldn't read from the FC when it's checked. I think Device Read Unprotect requested was not accepted well by the FC.
[image]

Am I doing something wrong? Please advise. 

Thank you,

SH

jhemcu, 2021-12-22:
STM32Cube  是需要ST-LINK 接SWD接口才能进行程序的烧写，GHF405AIO PRO 没有外留SWD端口，所以无法使用 STM32Cube.
 DfuSe Demo 3.0.6  安装完成后，还需要安装DFU的驱动程序，下载驱动的网址发送给你。
https://github.com/jhemcu/FC-ESC-Firmware/tree/main/Setup

董家良
津航电子

 联系我

francis916, 2021-12-22:
Thank you so much! I'll try it tonight and report back. 

Best,

SH

francis916, 2021-12-22:
Hi. I tried but DfeSe Demo 3.0.3 doesn't detect GHF405 AIO Pro; there's none in 'Available DFU Devices' still..

[image]

I installed DFU driver by clicking 'OS_Detect' in DFU Drivers. It's strange because Betaflight Configurator detects the DFU. I tried zadig but installing other drivers for "STM32 Bootloader" didn't help. WinUSB is the current one. I also tried https://ftdichip.com/drivers/ but it didn't help either. 

I guess there isn't much that I can do now? 

Thanks.

jhemcu, 2021-12-23:
建议换一台没有装过 zadig  驱动系统的电脑，不排除它和ST的DfeSe Demo 3.0.3 驱动冲突了！DfeSe Demo 3.0.3 驱动需要到设备管理器里面进行添加。

1.按键BOOT键，在连接USB线
2.在设备管理器里面添加*.RAR驱动

3.驱动安装成功会有检测到 如下

以上方法是属于软件应用，你可以先研究一下最终达到这个结果，我只能帮你到这里了。

董家良
津航电子

 联系我

francis916, 2021-12-23:
Thanks for the help. I'll try to use a computer that never installed zadig before. I think I'll reinstall Windows on the PC so it'll take some time. 

Thanks,

SH

francis916, 2021-12-23:
Hi. I tried Dfese Demo on another PC and installed the driver from the location that the program is installed. The program detected the FC in DFU. I tried DFU File Manager to create a dfu file from the hex file, and then downloaded it into the FC using program 'DFU Tester' after 'Erase' the FC. The Download went well and it showed 'Success' at the end. 

However, the FC still doesn't work. Betaflight configurator still can't connect to the FC. The red light is always on and not blinking even though it's not in DFU mode. 

I tried 'Upload' the FC content into DFU Tester and saved it into DFU file and then read it from the program. It's all filled with 'FF'. So the Download didn't actually work. 

Please let me know what you think. 

Thanks.

jhemcu, 2021-12-24:
根据你的描述，你已经成功完成了第一步HEX的烧录程序，你现在将飞控板连接到USB,在BETAFLIGHT里面可以看到COM口吗，如果可以，就连接进去，这个时候状态是没有陀螺仪的，需要你在CLI里面将我之前发给你的TXT文件里面的内容，全部复制并粘贴到CLI里面，按回车键，让飞控自动重启后，便完成了刷机工作。

董家良
津航电子

 联系我

jhemcu, 2021-12-24:
董家良
津航电子

 联系我

francis916, 2021-12-24:
Okay. I'll try to connect to Betaflight. Thanks.

francis916, 2021-12-24:
Hi. I tried to connect to Betaflight but I couldn't. I tried on PC and Mac. Would you be able to take a look if I ship it to JHEMCU office? Please send me an email with the shipping info if that can be done. 

Thanks,

SH

jhemcu, 2021-12-25:
邓克滔 15388113296, 四川省 成都市 金牛区 天回镇街道 北星大道一段6408号国际商贸城1号楼1402号（一定留纸条写清楚问题原因，联系方式）

董家良
津航电子

 联系我

jhemcu, 2021-12-25:
你目前是在中国吗 

董家良
津航电子

 联系我

francis916, 2021-12-25:
Nope. I'm shipping it from South Korea. :) It's gonna cost half the price of the FC but.. I wish to understand what's the problem because I bought another one and I want to know if I can try updating the FC..

Thanks.

jhemcu, 2021-12-26:
从韩国发到中国来维修，运费会很高，不合算。按我说的方法，是肯定可以更新固件的

董家良
津航电子

 联系我