jhemcu/FC-ESC-Firmware #4: Where is the unified target

sugaarK, 2021-12-31:
@jhemcu  where is your unified target for 4.3? This repo has been open for months...  we are about to release 4.3 release and your new h7 AIO doesn’t have a working target..  I know people buying these things and they can’t use them with the current code base...

blakemberry, 2022-01-05:
@sugaarK have you figure out a work around? Just put this board on my babytooth and am having alot of issues. IE only two motors spin up. All chime up when plugged into betaflight. If I disable rpm filtering all motors work properly. Any help with setup would be greatly appreciated.

jhemcu, 2022-01-05:
特别说明，由于H743固件的原因，暂时不支持双向DSHOT！

董家良
津航电子

 联系我

jhemcu, 2022-01-05:
特别说明，由于H743固件的原因，暂时不支持双向DSHOT！

你可以刷入  IFLIGHT_H743_AIO 固件

目前是BF平台的H743对双向DSHOT的支持都不好。包含所有品牌，请知晓

董家良
津航电子

 联系我

主题: [jhemcu/FC-ESC-Firmware] Where is the unified target  (Issue #4)

howels, 2022-01-29:
I am running the Matek H743 with bidirectional DShot. This is working correctly in BetaFlight 4.3.

Please submit an update so that JHEH743 can be used with latest BetaFlight 4.3.

I have helped earlier to create a unified target for your JHEF405-PRO, we would appreciate a unified target config from JHEMCU for JHEH743.  This looks like a fantastic board but it needs software for customers to buy it and enjoy the product.

howels, 2022-01-29:
I got bidirectional dshot working on all 4 motors with dshot_bitbang, so looks like a timer config issue.

sugaarK, 2022-01-30:
I agree @howels it the manufacturer needs a help with this we the Betaflight team can help. Also @jhemcu  know that after we finalise the 4.3 release we’ll be doing specifications for manufacturers to follow with new flight controllers so we can avoid these issues moving forward..  and questions @jhemcu  please email in English  at sugarK@betaflight.com