jhemcu/FC-ESC-Firmware #6: JHEMCU GF16 F405 problem OSD

waleri69, 2022-06-14:
JHEMCU GF16 F405 Flight Controller
Purchased a GF-16. INAV firmware version 5.0.0. I'm setting it up for now. There is a problem with OSD! The OSD works and is configured using the transmitters knobs. In INAV 5.0. I can't configure OSD!
INAV does not read OSD and does not configure ! I tried to change the font and change the type of TV signal. Nothing works out. Help fix this problem!!!
I went through Betalight and checked. OSD in Betaflight is configured both from the computer and from the remote control. There are no problems in Betaflight.
In INAV 5.0.0. OSD is not readable or configurable from the computer.
Nothing works out. Help fix this problem!!!

Even INAV 5.0 does not open the Black Box tab and does not read information from the black box !!! The MIXER tab doesn't open.

jhemcu, 2022-07-01:
Just now we generated the 6.0.0 firmware, and actually tested the OSD. You can download it from the email attachment. You can write the 6.0.0 firmware for testing. The download address of the assistant software is as follows:

http://seyrsnys.myzen.co.uk/inav-configurator-next/

主题: [jhemcu/FC-ESC-Firmware] JHEMCU GF16 F405 problem OSD (Issue #6)

jhemcu, 2022-07-08:
Let you wait for a long time, we found out the reason, because the previously provided beta firmware was caused. You are re-updating the firmware in my current attachment and it will be OK

主题: [jhemcu/FC-ESC-Firmware] JHEMCU GF16 F405 problem OSD (Issue #6)

jhemcu, 2022-10-11:
You go to this address to download the firmware for testing：https://github.com/jhemcu/FC-ESC-Firmware/blob/main/INAV/inav_5.0.0_JHEMCUF405.rar

主题: [jhemcu/FC-ESC-Firmware] JHEMCU GF16 F405 problem OSD (Issue #6)