jhemcu/FC-ESC-Firmware #7: JHEMCU GF16 F405

kvng1, 2022-07-20:
Can you please share the inav target.h & target.c? Need to know which pwm output are for motor or servo.
Any plane to build arduplane hex?
Thanks

jhemcu, 2022-07-21:
请查收

kvng1, 2022-07-23:
You did provide the hex file for flashing but there is no wiring instruction. I need to know which output for motor and which output for servo. That's why I am asking for target.h & target.c.

jhemcu, 2022-07-23:
I will send it to you on monday

kvng1, 2022-07-24:
> I will send it to you on monday
> […](#)

thanks

josemaria-vilaplana, 2023-07-02:
Any update on this? I also need inav target.h & target.c for GF16.
Thanks

jhemcu, 2023-07-03:
> Any update on this? I also need inav target.h &amp; target.c for GF16.

I will send you target. h&amp;target. c, as well as the compiled HEX.

josemaria-vilaplana, 2023-07-03:
> I will send you target. h&amp;target. c, as well as the compiled HEX.
> […](#)

Thanks

josemaria-vilaplana, 2023-07-14:
Any update on this? @jhemcu can you provide inav target.h & target.c for GF16?
Thanks

Strat00s, 2023-07-21:
I don't get this. @jhemcu you gus release binaries, yet releasing the board configuration (target.h and .c) would be much less work for you and much more useful for us (because we can then build our own binaries).

jhemcu, 2023-07-22:
I already sent you the C and H files last time
Please refer to the attachment

josemaria-vilaplana, 2023-07-23:
Has anyone received the inav target.h & target.c files? Can you share them?

kvng1, 2023-07-23:
Hi, he did send me last year, and I revised something to fit my drift build but I can't remember what changes.

https://drive.google.com/drive/folders/1qTyVCum27T9czTYmYijQ-6j5yrQMl5Qo?usp=sharing

NikolayBinom, 2025-04-11:
In Inav 7, everything was set up, however, after switching to inav 8, it is not possible to connect the receiver via sbus. I tried to connect to RX3 with inversion, the connection also does not happen. Is it possible to get the latest firmware for Inav 8 for GF16-F405?

jhemcu, 2025-04-11:
What I need to tell you is that SBUS is configured with UART1, not UART3. The firmware name that INAV needs to write is JHEMCUF405. The ports for the receiver are the same for INAV7 and INAV8.

NikolayBinom, 2025-04-11:
I know about uart1, I also tried using uart3. uart1 doesn't work. I assume that there is some kind of error in the 405 firmware.

jhemcu, 2025-04-12:
It works normally in INAV7, but not in INAV8. It should not have such obvious problems in INAV.