jhemcu/FC-ESC-Firmware #11: Telemetry doesn't work on GHF420AIO-BGA

anthropo, 2023-03-03:
https://github.com/betaflight/betaflight/issues/12442

I have R-XSR with ACCESS, and a JHEMCU GHF420AIOPro Board.
I tried Sbus and smartPort on different Uart TX, and also Fport on different UARts, and finally sBus and SoftSerial, smartPort and non inverted smartport, none of them worked, but receiver works (I can see orders in receiver tab).
I tried the same on aother board (mini MK2 Mamba) and my receivers work (R-XSR).

So my conclusion is that there is a bug with telemetry on GHF420AIO-BGA

To Reproduce :
Connect sbus to sbus pad, smartPort to TX3 pad, configure UART 6 to serial rx , UART 3 telemetry output to smartPort, and put your receiver options on smartport.
Check new devices on telemetry on the transmitter, nothing happens. Only BtRx, RSSI and VFR

Expected behavior : 
Telemetry devices should appear on the transmitter

jhemcu, 2023-03-04:
Have you updated the firmware? Is the current flight control firmware JHEF405? I will send the factory firmware to you, and you can rewrite it to AIO for testing. Because the underlying firmware of BF is the same, that is to say, the flight control function of F405 is the same for each brand. Please see the attachment

主题: [jhemcu/FC-ESC-Firmware] Telemetry doesn't work on GHF420AIO-BGA  (Issue #11)

 
betaflight/betaflight#12442

jhemcu, 2023-03-04:
You can send me the screenshots of the port configuration and receiver configuration in your BF, and also take pictures of the physical wiring to me。

主题: [jhemcu/FC-ESC-Firmware] Telemetry doesn't work on GHF420AIO-BGA  (Issue #11)

anthropo, 2023-03-04:
resource SERIAL_TX 3 none
resource SERIAL_RX 3 none
resource SERIAL_TX 11 B10
save

feature SOFTSERIAL
set serialrx_provider = SBUS
feature TELEMETRY

# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

[image]
[image]
[image]
[image]

jhemcu, 2023-03-04:
Our flight control doesn't need to change these contents. Please refresh the attachment firmware in my last email。I saw that the wiring was correct.

anthropo, 2023-03-04:
sorry i dont see any attachment in the mail

jhemcu, 2023-03-04:
I will send it to you once. Please check it。

anthropo, 2023-03-04:
what i just did :
Download https://github.com/jhemcu/FC-ESC-Firmware/tree/main/GHF420AIO-BGA(ICM-42688-P)
flashed the firmware and CLI the GHF420AIO-BGA.txt

then assigned Telemetry output on Smartport for UART3 and serialRx to UART6

No more joy, i receive orders from the transmitter but no telemetry back

jhemcu, 2023-03-04:
I see your previous screenshots. There seems to be no settings in this one. You can try it first. If it is not available or you can replace it with TX1 and TX2 to test the telemetry function. Remember to open the corresponding port function.

anthropo, 2023-03-05:
But I managed to have Telemetry via SoftSerial on LED_STIP

After that i meticulously tried to reproduce on TX1,2,3,4,6 but without result
Same for FPORT, i manage to receive but no telemetry.

Can you try by your side  on your board ? 

Thank you

jhemcu, 2023-03-09:
I'm glad you can solve this problem, but I don't understand why we need to change it to the LED pad. If the native UART port of BF doesn't work, there is a problem with the firmware of BF. Now the mainstream is ELRS receiver. We are not following up on FRSKY receiver. I'm very sorry.

anthropo, 2023-03-09:
I can't solve the problem, it is just a way to proove telemetry is working with a workaround, at the price of loosing Led strip. 
But I need led_strip too.
I don't buy a AIO FC to renounce to one of the possibility of the FC.
I'm very surprised with your answer. So you build and sell FC with no support ? I suppose sellers should know via social network ...

Please make an effort, catch a GHF420AIO-BGA from your factory and give a try with a R-XSR