jhemcu/FC-ESC-Firmware #14: IRC Tramp protocol on Tran 3016W

RandyReover, 2023-04-19:
I've trying to get IRC Tramp protocol work on Jhemcu Tran 3016, but it doesn't work.
So there are technical questions about Tramp implementation on this VTX:
1. Have you tested this VTX on BetaFlight/INav or ArduPilot? 
2. Could you provide with some dumps of IRC Tramp commands to your VTX? I'm trying to send following commands but the VTX just not answer (waiting about 3sec):
a) Init ('r' cmd): 0x0F, 0x72, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x81, 0x00
b) Get Config ('v' cmd): 0x0F, 0x76, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x85, 0x00

Commands sent by halfduplex UART with [9600bps, 8N1, not inverted, 5V logic level] characteristics, after command sent, uart is listening, but there is no any byte of response.
I'm able to provide Arduino project (tested on ArduinoMega2560Pro) to make tests, this project succesfully works with Eachine TX5258(in SmartAudio mode).
On Betaflight forums I've read about 'poor implementation of IRC Tramp and SmartAudio protocols', there are some workarounds with AKK/RDQ VTX's, but it doesn't work with Tran3016W.
So, could you provide working dumps of commands (or working code examples) to this VTX to get it switch power and channels by wire?

jhemcu, 2023-04-19:
I have sent you the JSON file. Please refer to the attachment。Please do not make any modifications to the JSON file on your own, as this may prevent you from using image transmission correctly. You do not need to question whether it has been tested on BF and INAV, as this is necessary. If it cannot be used normally, why do we dare to ship in bulk. In addition, our product information can be downloaded from this website, or you can ask your seller for information.

 https://github.com/jhemcu/FC-ESC-Firmware

主题: [jhemcu/FC-ESC-Firmware] IRC Tramp protocol on Tran 3016W  (Issue #14)

RandyReover, 2023-04-22:
So now I'm able to confirm that IRC Tramp correctly works on JHEMCU Ruibet Tran3016W.
There was a problem with crc calcultation (there were difference between ArduPilot and Betaflight crc calculation, and ArduPilot procedure is correct (calc from second to 13th byte, crc is written to 14th byte)).
So, correct Init (0x72, 'r') command command dump:
0F 72 00 00 00 00 00 00 00 00 00 00 00 00 72 00
Received response: 
0F 72 EC 13 70 17 58 02 00 00 00 00 00 00 52 00
Correct GetConfig (0x76, 'v') command dump:
0F 76 00 00 00 00 00 00 00 00 00 00 00 00 76 00
Received response:
0F 76 78 16 19 00 00 00 00 00 00 00 00 00 1D 00

closed 2023-04-22 by RandyReover